manio/skymax-demo #6: Invalid JSON in hex stream

cowpen, 2020-02-22:
I have an MPP Solar LV 5048. It's a split phase unit containing two inverters and two charge controllers. It shares the same comms protocol as the supported models listed here, although it supports more query & result codes (i,e, QPIGS2 for the second phase, etc.). I'm using node-red to call skymax-demo and injecting stdout to influxdb. The flow is successful, but the output returned contains an 0x0 in the buffer stream which the function to parse to a JSON object chokes on. The location of the 0x0 in the output string is at the value location for "AC_charge_on:,"

[image]

I've looked through the skymax code but I don't really have a clue for what I'm looking for. Could it be related to the "-" in line 202 of main.cpp?

Thanks for your help!

manio, 2020-02-24:
The problem is in the line:
`"AC_charge_on":,`
It is wrong formatted JSON.

cowpen, 2020-02-24:
Ahh... thank you. Looking more carefully, I see now that the QPIGS protocol for my inverter is a bit different for the parameter response at position "x".  That must be where the bad JSON is coming from. 
[LV5048 5KW protocol-20190222 for customer-1.pdf](https://github.com/manio/skymax-demo/files/4244304/LV5048.5KW.protocol-20190222.for.customer-1.pdf)

dansass, 2020-04-15:
hey cowpen! I am at this stage to try to talk to my inverter with node red. Would u do me a favor and post your flow file that works? Would be greatly appreciated. to do that, click the hamburger menu than "export". Thanks

cowpen, 2020-04-15:
Here's what I had when I gave up on this particular solution since it didn't support the queries and responses required for my inverter. Hope it's helpful in some way.
[flows(1).zip](https://github.com/manio/skymax-demo/files/4481615/flows.1.zip)

Edgepath, 2023-06-27:
Hello, 

trying to get this working with MasterPower (also Voltronic) [MasterPower PMS 5K-48TOP Off-Grid](https://duckduckgo.com/?t=ffab&q=MasterPower+PMS+5K-48TOP+Off-Grid)

HINT: does this also work with USB-to-USB-Cable (Voltronic internal USB to Serial Cypress adapter) or only with COM-JR45 connection? :D

what is the content of /home/pi/inverter.sh ? :D