manio/skymax-demo #8: cmake error

dansass, 2020-10-04:
when i wnat to run cmake .. && make it says

CMake Error: The source directory "/home/pi" does not appear to contain CMakeLists.txt.

and i am in the skymax-demo directory
 
whats going on?

manio, 2020-10-04:
You are not in skymax-demo directory :)

dansass, 2020-10-05:
I think I am

[image]

manio, 2020-10-07:
you're right
just type cmake (without ..) and see if it helps

dansass, 2020-10-07:
Still no luck :( 
here are my commands that i ran

[image]

[image]

[image]

manio, 2020-10-09:
what if you do:
```
clone [...]
cd skymax-demo
mkdir build
cd build
cmake ..
make
```

dansass, 2020-10-09:
[image]

Im wondering if i Maybe installed cmake wrong. It sure is acting weird

manio, 2020-10-09:
Yeah, you probably have something wrong.
Try to clone to some other dir then your home dir, like /tmp
For me it is working no matter if i do "cmake . " in main dir, or "cmake .." in build subdir:
```
/tmp/skymax-demo/build/ cmake ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/skymax-demo/build
```

manio, 2023-03-24:
This is already covered in https://github.com/manio/skymax-demo/issues/17
Closing this three-years old issue.

closed 2023-03-24 by manio