manio/skymax-demo #11: inverter_poller does not open the device=

dilyanpalauzov, 2022-03-07:
My /etc/inverter/inverter.conf contains:

device=/dev/hidraw0
run_interval=1
amperage_factor=1.0
watt_factor=1.01

Calling `stdbuf -e0 -o0 strace -s2048 ./inverter_poller -d |&tee e` contains [e.txt](https://github.com/manio/skymax-demo/files/8198332/e.txt).

As can be seen, inverter_poller does not open the device.  As can be seen, the device /dev/hidraw0 is not opened and the output contains `INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 48.0 44.0 59.5 54.7 2 02 100 0 1 3 9 01 0 1 54.0 0 1 0 00014 000`.

manio, 2022-03-07:
I know that different inverters has some more or less differences in the reply frame. From what I can see the inverter is properly opened, and properly read, because it is outputting the correct frame:
```(230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 48.0 44.0 59.5 54.7 2 02 100 0 1 3 9 01 0 1 54.0 0 1 0 00014 000```
So it is responding properly for a query. The problem is probably in some different thing, or protocol differences. I know that even the same inverter may respond differently depending on firmware version, not saying about different devices/clones.

I have four tips for you:
1. if you have some programming skills: try to see why it is not parsing the data correctly (there is probably some small adjustment in the code which need to be added; I would start from removing start/stop sequence checker)
2. try https://github.com/ned-kelly/docker-voltronic-homeassistant - as you can see in the _readme_ - it is a fork which is better maintained. My project was rather a demo/proof-of-concept.
3. try to revert @ned-kelly changes on my project, typing `git checkout 48826f4` in the project root directory, rebuild and check again
4. if you only need to pass the data from your inverter to the _grafana_, then you may also consider using [hard](https://github.com/manio/hard). You can then disable all unnecessary modules and set it to use _skymax_ and _grafana_ only. Configuration file can then be similar to this: https://skyboo.net/2021/06/huawei-sun2000-monitoring/
This approach however only supports *my* inverter and *my* specific protocol - I am not willing to add more inverters support there as this is a very customized software.

dilyanpalauzov, 2022-03-07:
Thanks for your answer.  It was a matter of setting `qpiri=102` in inverter.conf.