manio/skymax-demo #23: "incorrect start/stop bytes" with Voltacon Conversol max 8kW off-grid inverter

A13xund3r, 2023-10-29:
Inverter: https://voltaconsolar.com/off-grid/conversol-max-8kw-off-grid.html
Finally I've got this working and this id output with debug option:
`bananapi:skymax-demo:# ./inverter_poller -d                                                                                                                                                     
Sun Oct 29 14:33:23 2023 INVERTER: Debug set
Sun Oct 29 14:33:23 2023 INVERTER: Current CRC: 49 C1
Sun Oct 29 14:33:23 2023 INVERTER: QMOD reply size (5 bytes)
Sun Oct 29 14:33:23 2023 INVERTER: QMOD: 5 bytes read: (L
Sun Oct 29 14:33:23 2023 INVERTER: QMOD query finished
Sun Oct 29 14:33:23 2023 INVERTER: Current CRC: B7 A9
Sun Oct 29 14:33:24 2023 INVERTER: QPIGS reply size (110 bytes)
Sun Oct 29 14:33:24 2023 INVERTER: QPIGS: 110 bytes read: (239.5 49.9 239.5 49.9 0742 0631 009 375 38.60 000 000 0040 00.9 101.5 00.00 00000 00010110 00 00 00099 010
Sun Oct 29 14:33:24 2023 INVERTER: QPIGS query finished
Sun Oct 29 14:33:24 2023 INVERTER: Current CRC: F8 54
Sun Oct 29 14:33:24 2023 INVERTER: QPIRI reply size (98 bytes)
Sun Oct 29 14:33:24 2023 INVERTER: QPIRI: incorrect start/stop bytes.  Buffer: (230.0 34.7 230.0 50.0 34.7 8000 8000 48.0 44.0 44.0 56.4 54.0 0 002 020 0 1 3 9 01 0 0 54.0 0 1 400099 010
Sun Oct 29 14:33:24 2023 INVERTER: Current CRC: B4 DA
Sun Oct 29 14:33:24 2023 INVERTER: QPIWS reply size (36 bytes)
Sun Oct 29 14:33:24 2023 INVERTER: QPIWS: incorrect start/stop bytes.  Buffer: (000000000000000001000000000000010000 48.0 44.0 44.0 56.4 54.0 0 002 020 0 1 3 9 01 0 0 54.0 0 1 400099 010
`
Without parameters nothing is thrown to the console. Different comm protocol?

manio, 2023-10-30:
Hi,
It is normal that the protocol differs slightly.
Thanks to @chadek we now have the parameters now merged (https://github.com/manio/skymax-demo/pull/22).
Isn't it sufficient to set it in the ini ?

chadek, 2023-10-30:
[Here](https://watts247.com/manuals/mpp/PIP-HS_MS_MSX_HYBRIDV%20RS232%20Protocol-C_20170822_.pdf) is a slightly more recent version of the protocol. If compare with the 2014 version, it seams that they did not change the wall thing but only some more information to the protocol.

manio, 2023-10-30:
> Maybe we could add some comments into the example config file (and why not into the read me)?

I am open to PR :)

chadek, 2023-10-30:
@A13xund3r I might have found a doc that could match your model protocol version : [this one](https://domosimple.eu/wp-content/uploads/2022/10/MAX-Communication-Protocol-for-HV7.2k-LV5k-V00-20200717.pdf) seams to have a QPIGS2 query that supposedly would output your MPPT 2 value.

A13xund3r, 2023-10-30:
> @A13xund3r I might have found a doc that could match your model protocol version : [this one](https://domosimple.eu/wp-content/uploads/2022/10/MAX-Communication-Protocol-for-HV7.2k-LV5k-V00-20200717.pdf) seams to have a QPIGS2 query that supposedly would output your MPPT 2 value.

Something is wrong with raw command receiver, it always read 7 bytes but apparently you are correct, I need QPIGS2 command
```
bananapi:skymax-demo:# ./inverter_poller -d -r QPIGS                                                                                                                                           
Mon Oct 30 11:40:37 2023 INVERTER: Debug set
Mon Oct 30 11:40:37 2023 INVERTER: Current CRC: B7 A9
Mon Oct 30 11:40:37 2023 INVERTER: QPIGS reply size (7 bytes)
configuration options for the actual inverter polling process...Buffer: (240.7 

bananapi:skymax-demo:# ./inverter_poller -d -r QPIGS2                                                                                                                                          
Mon Oct 30 11:43:34 2023 INVERTER: Debug set
Mon Oct 30 11:43:34 2023 INVERTER: Current CRC: 68 2D
Mon Oct 30 11:43:34 2023 INVERTER: QPIGS2 reply size (7 bytes)
onfiguration options for the actual inverter polling process...  Buffer: (00.0 0-
```
Currently second PV is not connected so 0V is expected.
```
bananapi:skymax-demo:# ./inverter_poller -d -r QMOD                                                                                                                                            
Mon Oct 30 11:44:27 2023 INVERTER: Debug set
Mon Oct 30 11:44:27 2023 INVERTER: Current CRC: 49 C1
Mon Oct 30 11:44:30 2023 INVERTER: QMOD read timeout
Mon Oct 30 11:44:30 2023 INVERTER: QMOD reply too short (5 bytes)
Reply:  
```
reply line is always empty.
I'm happy to test but I'm assembler/basic hobbyist programmer so fiddling with C++ is extremely difficult for me.

chadek, 2023-10-30:
@A13xund3r you should be able to query your inverter with the code on [this PR](https://github.com/manio/skymax-demo/pull/24), using the new -l option to set the length of your buffer. Good luck with the length guessing!

chadek, 2023-10-30:
> > Maybe we could add some comments into the example config file (and why not into the read me)?
> 
> I am open to PR :)

I'll open one to update config and the readme, including the buffer length values I just removed on [PR](https://github.com/manio/skymax-demo/pull/24)

A13xund3r, 2023-10-30:
> @A13xund3r you should be able to query your inverter with the code on [this PR](https://github.com/manio/skymax-demo/pull/24), using the new -l option to set the length of your buffer. Good luck with the length guessing!

It works, no need to guess, I can count bytes from PDF you found as my inverter is using this protocol or set something higher and let program tell me message length:
```
bananapi:skymax-demo:# ./inverter_poller -l 160 -d -r QPIGS2
Mon Oct 30 22:02:39 2023 INVERTER: Debug set
Mon Oct 30 22:02:39 2023 INVERTER: Current CRC: 68 2D
Mon Oct 30 22:02:42 2023 INVERTER: QPIGS2 read timeout
Mon Oct 30 22:02:42 2023 INVERTER: QPIGS2 reply too short (21 bytes)
Reply:
bananapi:skymax-demo:# ./inverter_poller -l 21 -d -r QPIGS2
Mon Oct 30 23:13:19 2023 INVERTER: Debug set
Mon Oct 30 23:13:19 2023 INVERTER: Current CRC: 68 2D
Mon Oct 30 23:13:19 2023 INVERTER: QPIGS2 reply size (21 bytes)
Mon Oct 30 23:13:19 2023 INVERTER: QPIGS2: 21 bytes read: (00.0 000.0 00000
Mon Oct 30 23:13:19 2023 INVERTER: QPIGS2 query finished
Reply:  00.0 000.0 00000
```
After -l 160 inverter was replying 7 bytes (NAK to every command until i started slsnif sniffer to see what is going on

A13xund3r, 2023-11-01:
After hours of googling it seems my inverter use [PI30MAX protocol](https://github.com/jblance/mpp-solar/blob/master/docs/protocols/PI30MAX.Communication.Protocol20210217.pdf). So far everything does match with my inverter.

chadek, 2023-11-01:
I don't have access to my hardware to test this out, but it should solve this issue when it is done : https://github.com/manio/skymax-demo/pull/26

chadek, 2023-11-17:
@A13xund3r I just found out that the document you point out describing the PI30MAX protocol, is part of a wonderful repo supporting a lot of inverter protocol, including yours. You should try it, it seams to work quite well :)

A13xund3r, 2023-11-17:
It is in python, not my preferred language to run on SBC https://www.freecodecamp.org/news/python-vs-c-plus-plus-time-complexity-analysis/ but i have it on my list to check. Just day is too short...

A13xund3r, 2024-04-24:
Closing as it is fixed.

closed 2024-04-24 by A13xund3r