manio/skymax-demo #27: Zero SCC-voltage

melnik-cto, 2024-02-01:
Hello!
New problem with  Voltronic Axpert King II.
As I can see in your code, pv_input_watts = (scc_voltage * pv_input_current) * wattfactor;
But SCC voltage always returns 0.  Accordingly, there is no real way to estimate how much the inverter actually receives through solar panels.
Example of output:
(
    [Inverter_mode] => 3
    [AC_grid_voltage] => 225.3
    [AC_grid_frequency] => 50
    [AC_out_voltage] => 228.1
    [AC_out_frequency] => 49.9
    [PV_in_voltage] => 275.4
    [PV_in_current] => 1
    [PV_in_watts] => 0
    [PV_in_watthour] => 0
    [SCC_voltage] => 0
    [Load_pct] => 15
    [Load_watt] => 662
    [Load_watthour] => 22.0667
    [Load_va] => 775
    [Bus_voltage] => 407
    [Heatsink_temperature] => 49
    [Battery_capacity] => 100
    [Battery_voltage] => 53.9
    [Battery_charge_current] => 5
    [Battery_discharge_current] => 0
    [Load_status_on] => 1
    [SCC_charge_on] => 1
    [AC_charge_on] => 1
    [Battery_recharge_voltage] => 46
    [Battery_under_voltage] => 42
    [Battery_bulk_voltage] => 56.5
    [Battery_float_voltage] => 54
    [Max_grid_charge_current] => 20
    [Max_charge_current] => 30
    [Out_source_priority] => 1
    [Charger_source_priority] => 0
    [Battery_redischarge_voltage] => -0.4
    [Warnings] => 000000000000000000000000000000000000
)

I saw your comment about pv_input_current, and try change scc_voltage to PV_in_voltage, but pv_input_current return integer value, but the power graph is very step-like. At least a little closer to reality.so 

Example on image.

Do you have any idea how fix this? Thanks
[image]

manio, 2024-02-03:
I think that on your graph the `PV_in_current` is returing value "1" when `PV_in_watts` started generating (seems like this is 0 - not generating/1-generating boolean value here). And when it occur you see on graph that it is in pair with `PV_in_voltage`. I don't know how can I further help, especially because I don't have any PV connected to my inverter.

melnik-cto, 2024-06-12:
@manio hello. I found solution for my case.
Raw QPIGS data seem like that:
`219.2 49.9 229.9 49.9 0897 0688 017 403 53.40 004 100 0053 0001 232.9 00.00 00000 00010111 00 00 **00240** 110 0 00 0120`
If we try compare with what we get, then:
[image]

So, I found funny solution and it works for me (Voltronic Axpert King II). I modified your main.cpp
`                sscanf(reply1->c_str(), "%f %f %f %f %d %d %d %d %f %d %d %d %f %f %f %d %s %d %d %d", &voltage_grid, &freq_grid, &voltage_out, &freq_out, &load_va, &load_watt, &load_percent, &voltage_bus, &voltage_batt, &batt_charge_current, &batt_capacity, &temp_heatsink, &pv_input_current, &pv_input_voltage, &scc_voltage, &batt_discharge_current, &device_status, &garbage1, &garbage2, &pv_power);
`