manio/skymax-demo #29: Can't read data from Voltronic AXPERT MKS IV 5.6KW

sebashb, 2024-07-23:
I have tried to connect to a Voltronic AXPERT MKS IV 5.6KW but I get the following error message `INVERTER: Unable to open device file (errno=2 No such file or directory)`

```
user@device:~/skymax-demo$ lsusb
Bus 006 Device 002: ID 05e3:0620 Genesys Logic, Inc. 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 041: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 005 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 13d3:3548 IMC Networks 
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

user@device:~/skymax-demo$ ls /dev/hid*
/dev/hidraw0

user@device:~/skymax-demo$ sudo ./inverter_poller -d
Tue Jul 23 22:30:20 2024 INVERTER: Debug set
Tue Jul 23 22:30:20 2024 INVERTER: Current CRC: 49 C1
Tue Jul 23 22:30:23 2024 INVERTER: QMOD read timeout
Tue Jul 23 22:30:23 2024 INVERTER: QMOD command timeout, or couldn't find stop b configuration options for the actual inverter polling process...

# The device to read from...
# Use: /dev/ttyS0 if you have a serial device,
#      /dev/ttyUSB0 if a USB<>Serial,
#      /dev/hidraw0 if you're connecting via the USB port on the inverter.

device=/dev/hidraw0

# How many times per hour is the program going to run...
# This is used to calculate the PV & Load Watt Hours between runs...
# If unsure, leave as default - it will run every minute...

# (120 = every 30 seconds)...
run_interval=120

# This allows you to modify the amperage in case the inverter is giving an incorrect
# reading compared to measurement tools.  Normally this will remain '1'
amperage_factor=1.0

# This allos you to modify the wattage in case the inverter is giving an incorrect
# reading compared to measurement tools.  Normally this will remain '1'
watt_factor=1.01

# Depending on the protocol version used and your device,
# you might need to change the following parameters

#### For Axpert VM3 inverter like, use tho 
Tue Jul 23 22:30:23 2024 INVERTER: Unable to open device file (errno=2 No such file or directory)
Tue Jul 23 22:30:28 2024 INVERTER: Current CRC: F8 54
Tue Jul 23 22:30:31 2024 INVERTER: QPIRI read timeout
Tue Jul 23 22:30:31 2024 INVERTER: QPIRI command timeout, or couldn't find stop byte. Byte read (32 bytes). Buffer: (230.0 24.3 230.0 50.0 24.3 5600r the actual inverter polling process...
```

I'm aware that I may need to change a few values on the `inverter.conf` file but I don't know how do I find the correct values.

manio, 2024-07-24:
Hi, it doesn't hold together:
you have the needed device:

> user@device:~/skymax-demo$ ls /dev/hid*
/dev/hidraw0

in config you have a proper value:
> device=/dev/hidraw0

But when starting you've got:
> Unable to open device file (errno=2 No such file or directory)

Are you sure that the config is OK and from the same instance?

sebashb, 2024-07-24:
Hi, yes I'm sure that I'm running inverter_poller from the same directory on which the configuration is at. 

Is there any simpler test that I can do to verify the proper function of the device? 

Could it be a driver issue? Or in that case the device wouldn't show up?

manio, 2024-07-24:
All seems fine, you have the needed device (/dev/hidraw0) but it seems, that `inverter_poller` cannot open it (or it is trying to open quite different file)?! weird!

Read my blog: https://skyboo.net/2017/03/monitoring-voltronic-power-axpert-mex-inverter-under-linux/

A13xund3r, 2024-08-09:
Lack of permission? Try root.

sebashb, 2024-08-13:
I'm running it as root, as shown on the console output I posted on the original question. 

I also noticed that after failing to run `inverter_poller` the Cypress Semiconductor USB to Serial disappears from `lsub` and the device also disappears from `/dev/hidraw0`.

A13xund3r, 2024-08-13:
> I'm running it as root, as shown on the console output I posted on the original question.

It should say root@ not user@ except if you heavily modified your system.

> I also noticed that after failing to run `inverter_poller` the Cypress Semiconductor USB to Serial disappears from `lsub` and the device also disappears from `/dev/hidraw0`.

This is symptom of faulty USB device, wrong driver or unspecified system issue (chipset, USB host, kernel).