manio/skymax-demo #30: Incorrect data handling in NAK response case

A13xund3r, 2024-11-16:
This program was working flawlessly until I switched inverter in to battery charging only for winter when it produce less energy than inverter itself consume. At some point I've found it feed impossible values to emoncms. Rebooting the inverter fixed it so I blamed inverter and ignored the issue but now it happened again and rebooting doesn't fix it. For sure something is wrong with the inverter as it reply NAK to every command but apparently program doesn't handle such case, only warnings section contain copy of the inverter response.
```
jesus:skymax-demo:# ./inverter_poller -1 -d
Sat Nov 16 15:29:21 2024 INVERTER: Debug set
Sat Nov 16 15:29:21 2024 INVERTER: Current CRC: 49 C1
Sat Nov 16 15:29:22 2024 INVERTER: QMOD reply size (7 bytes)
Sat Nov 16 15:29:22 2024 INVERTER: QMOD: 7 bytes read: (NAK
Sat Nov 16 15:29:22 2024 INVERTER: QMOD query finished
Sat Nov 16 15:29:22 2024 INVERTER: Current CRC: B7 A9
Sat Nov 16 15:29:22 2024 INVERTER: QPIGS reply size (7 bytes)
Sat Nov 16 15:29:22 2024 INVERTER: QPIGS: 7 bytes read: (NAK
Sat Nov 16 15:29:22 2024 INVERTER: QPIGS query finished
Sat Nov 16 15:29:22 2024 INVERTER: Current CRC: F8 54
Sat Nov 16 15:29:22 2024 INVERTER: QPIRI reply size (7 bytes)
Sat Nov 16 15:29:22 2024 INVERTER: QPIRI: 7 bytes read: (NAK
Sat Nov 16 15:29:22 2024 INVERTER: QPIRI query finished
Sat Nov 16 15:29:22 2024 INVERTER: Current CRC: B4 DA
Sat Nov 16 15:29:22 2024 INVERTER: QPIWS reply size (7 bytes)
Sat Nov 16 15:29:22 2024 INVERTER: QPIWS: 7 bytes read: (NAK
Sat Nov 16 15:29:22 2024 INVERTER: QPIWS query finished
INVERTER: ampfactor from config is 1.00
INVERTER: wattfactor from config is 1.01
{
  "Inverter_mode":0,
  "AC_grid_voltage":-0.0,
  "AC_grid_frequency":-0.4,
  "AC_out_voltage":-0.0,
  "AC_out_frequency":0.0,
  "PV_in_voltage":-0.0,
  "PV_in_current":0.0,
  "PV_in_watts":0.0,
  "PV_in_watthour":0.0000,
  "SCC_voltage":0.0000,
  "Load_pct":0,
  "Load_watt":1,
  "Load_watthour":0.0333,
  "Load_va":-1225116224,
  "Bus_voltage":1,
  "Heatsink_temperature":-1226581524,
  "Battery_capacity":-1225119904,
  "Battery_voltage":-0.00,
  "Battery_charge_current":-1225118896,
  "Battery_discharge_current":-1225120368,
  "Load_status_on":▒,
  "SCC_charge_on":,
  "AC_charge_on":,
  "Battery_recharge_voltage":-0.0,
  "Battery_under_voltage":-0.0,
  "Battery_bulk_voltage":-0.4,
  "Battery_float_voltage":-0.4,
  "Max_grid_charge_current":-1225120368,
  "Max_charge_current":0,
  "Out_source_priority":-1226567700,
  "Charger_source_priority":25442486,
  "Battery_redischarge_voltage":-0.4,
  "Warnings":"NAK"
}
```

manio, 2024-11-16:
NAK means "Not Acknowledge" so the inverter doesn't understand the request send by the program (or it was screwed during transport).

A13xund3r, 2024-11-17:
Ok, found why it is NAKing me, my bad habits took revenge on me. Still main issue is present, sure it should not happen (NAK response) in the first place but IMHO it need to be handled.

manio, 2024-11-17:
I don't got it... so what is the issue?

I can see that it is NAKing to every command you sent (QMOD, QPIGS, QPIRI and QPIWS). This is unusual.
Was it working before? There was not any recent changes here in this project.

And how is this supposed to be handled? The NAK is an error - the inverter doesn't respond with data, so don't expect to parse the data in this case.

A13xund3r, 2024-11-17:
After I switched inverter in to battery charger only mode I was getting inverter mode 0 reply because apparently your inverter don't have reply "C" to QMOD query. So I added another case to have it covered (will issue pull request later) but I did edit in local copy of your apparently old repo before [communication issue was fixed](https://github.com/manio/skymax-demo/pull/26) so it was working fine until I rebooted the SBC. This is entirely my fault of not keeping proper version control.

A13xund3r, 2024-11-18:
If you don't see need to handle NAK response - yes.

manio, 2024-11-18:
This project as a name is telling is a demo project (I know it grew up over time) but it is a low prio for me, as I am rather using [rust component](https://github.com/manio/hard/blob/master/src/skymax.rs) for this (only for my inverter) on "production".

But ofc I am open to pull requests :)

A13xund3r, 2024-11-18:
I know, you say that everywhere, if I will be able to add NAK case handling I will issue pull request.

closed 2024-11-18 by A13xund3r